kekyo/FlashCap #9: On Linux at least cant close than reopen Camera

tesses50, 2022-05-16:
An exception of type 'System.ArgumentException' occurred in System.Private.CoreLib.dll but was not handled in user code: 'FlashCap: Couldn't set video format [3]: DevicePath=/dev/video0'
   at FlashCap.Devices.V4L2Device..ctor(String devicePath, VideoCharacteristics characteristics, Boolean transcodeIfYUV, FrameProcessor frameProcessor)
   at FlashCap.Devices.V4L2DeviceDescriptor.OpenWithFrameProcessorAsync(VideoCharacteristics characteristics, Boolean transcodeIfYUV, FrameProcessor frameProcessor)
   at FlashCap.CaptureDeviceDescriptorExtension.OpenAsync(CaptureDeviceDescriptor descriptor, VideoCharacteristics characteristics, PixelBufferArrivedDelegate pixelBufferArrived)
   at MainForm.<SetCamera>d__18.MoveNext() in /home/mike/Desktop/TimelapseNow/Timelapse.Desktop/Program.cs:line 294

Works otherwise (as far as I can see)

kekyo, 2022-05-16:
It appears that the specified format could not be applied to the device.

* What did you choice for video characteristics?
  [Can I see all the fields of this type?](https://github.com/kekyo/FlashCap/blob/main/FlashCap/VideoCharacteristics.cs#L32)
* Could you dump your camera informations by `v4l2-ctl` ?
  For example:

```bash
kouji@phenex:~$ sudo v4l2-ctl --all

Driver Info:
	Driver name      : uvcvideo
	Card type        : Logitech Webcam C930e
	Bus info         : usb-0000:00:14.0-4.4
	Driver version   : 5.13.19
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : Logitech Webcam C930e
	Serial           : 15EB897E
	Bus info         : usb-0000:00:14.0-4.4
	Media version    : 5.13.19
	Hardware revision: 0x00000013 (19)
	Driver version   : 5.13.19
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : Logitech Webcam C930e
	Function         : V4L2 I/O
	Flags         : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000022: from remote pad 0x100000a of entity 'Processing 3': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV' (YUYV 4:2:2)
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 614400
	Colorspace        : sRGB
	Transfer Function : Rec. 709
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2000 max=7500 step=1 default=4000 value=4000 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=128
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
				1: Manual Mode
				3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=250 value=250 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=255 step=5 default=0 value=0 flags=inactive
                     focus_auto 0x009a090c (bool)   : default=1 value=1
                  zoom_absolute 0x009a090d (int)    : min=100 max=400 step=1 default=100 value=100

```

kekyo, 2022-05-16:
ああ、私は読み違いをしていたかもしれません。同じデバイスに対して最初のオープンは成功し、クローズして、再びオープンした場合、という事で合っていますか？

tesses50, 2022-06-17:
For me opening camera first time works (If I close and reopen camera, without restarting program, It crashes when I open it again)

kekyo, 2022-11-19:
@tesses50 Avalonia sample code has capable to switch devices and reproduce the reopen situation. [Modification added.] (8bcf2a54b85a822c86bc36b8367a06a142580d25) How about this?

kekyo, 2022-11-23:
1.4.0 merged and released.

closed 2022-11-23 by kekyo