kekyo/FlashCap #15: How to detect if camera is being used by other applications?

geocine, 2022-10-20:
The only method I could think off is this pseudocode. Is there a more reliable/better way to do this?
```
start camera
start timer to show camera is in use

OnPixelBufferArrivedAsync
   stop timer to show camera is in use
```

kekyo, 2022-10-28:
Sorry for the delay .

In the current version, there is no direct way to check if it is open or not. (If an error is detected, it is possible that an exception will be raised.)
Are you saying that if the camera device is already in use, it will open successfully but not call back?

Maybe there is a mistake in your error checking, or perhaps you need to check in a different way...
Is your environment Windows or Linux?

Translated with www.DeepL.com/Translator (free version)

geocine, 2022-10-31:
> Are you saying that if the camera device is already in use, it will open successfully but not call back?

Correct, I'm on Windows. I'm using the demo in this repo to reproduce issue.

kekyo, 2022-11-13:
Memoized: https://stackoverflow.com/questions/8655951/get-webcam-status-using-c-sharp

kekyo, 2022-11-14:
(This report contains with #16 )

I was able to reproduce your problem and have investigated. Unfortunately, it seems that the DirectShow filter graph cannot fully determine that you have entered this state.

The above memoized claims that it can be detected by having the filter graph play and examining its status. However, when I run playback with the capture device bound, DirectShow becomes unresponsive at that point.
Later, when I checked the status, I found that there is the following division of cases:

* After execution of FlashCap, another application uses the same camera: Filter graph status is continue "playing".
* Before execution of FlashCap, another application uses the same camera: Filter graph status is "stopped".

In the latter case, it is possible to recover (if another application stops) by monitoring it with an interval timer, etc.
And forcing the filter graph to stop and then play. However, in the end, I don't think this hack is appealing because it doesn't solve the former problem.

更に、他のアプリケーション（例えばWindows Camera、多分UWPベース）がどのように回避しているのかを少し調べました。それらのアプリケーションでは、少なくともDirectShowではない方法を使っているようです。
もっとも古く、しかしDirectShowより新しいMedia Foundation APIでは、エラーコードが定義されているのを確認しました。恐らくはUWPなどのハイレベルAPIでは、これを下敷きにしているのでしょう。

geocine, 2022-11-15:
Thank your for investigating

kekyo, 2022-11-16:
I too find this issue very disappointing, and I find the API design that does not attempt to address it offensive, considering that even though DirectShow is an old technology, it is still used in a wide variety of applications ☹

yangjieshao, 2023-02-01:
事实上 在实际使用时在启动时就判断设备是否被占用意义并不大
只需要由前端程序判断 在启动相机后连续3~5帧的时间内没有帧数据返回就可以认为相机异常了
异常包括被占用或者播放到一半相机被移除

3~5帧的时间非常短暂，对于用户来说可能也就是眨个眼的时间，前端页面甚至连连接中的动画都不用做

kekyo, 2023-02-02:
Thanks for the useful suggestions!