kekyo/FlashCap #29: How everyone is using it?

kekyo, 2022-12-09:
This issue is purely due to my interest. How do you use this library? Please tell me as much as you are comfortable with. Being able to know the situsations makes me realize that we have a growing user base, motivates me to develop more, and makes dinner taste better! 😄

yangjieshao, 2022-12-28:
中国正在推进政府操作系统的Linux化，有很多程序需要使用到usb摄像头，暂时为止我只找到了这一个c#在Linux连接usb摄像头的方式。
所以说，十分感谢，帮了大忙。

kekyo, 2022-12-28:
@yangjieshao Thank you! 谢谢!

shiggityshaggs, 2022-12-30:
I am using it with Tesseract for OCR. Thanks for the work you have put into this!
It's working with Logitech C920 and also the camera on the HTC Vive (Windows).
May I suggest opening a [Discussions forum](https://docs.github.com/en/discussions) unless you prefer Issues?

kekyo, 2023-01-01:
@shiggityshaggs Thanks for using FlashCap, and about the GH Discussion recommendations! I know about GH Discussion, but not enough to fill an issue at the moment, so I'll change that when the time comes 😄

sappho192, 2023-01-04:
Hello from South Korea and many thanks for this amazing repository!

For me, I made a demo WPF app which captures frame images from a webcam and record the audio via microphone.
Those images and audio file is transferred to deep learning API server which infers emotion from the human face and voice, so the WPF app will display current user's emotion state in real-time.

Before the FlashCap when I was undergraduate and M.S student, I always had to use the heavy and huge OpenCVSharp library only to acquire each frame data from the webcam to apply the postprocessing job. Perfermance was so dissatisfying but there was no other choice. But now I can use FlashCap!

Today I've found out the updates on the README.md of this repo and very surprised of your detailed descriptions and guidelines to properly programming the camera capture feature. I've been looking for those informations on the web but couldn't acquire that much...

It would be nice if you support the macOS (but personally not recommended) because there were some demands when I making the demo app.

From this week I'm participating another project in different organization but making pretty same thing and I'm going to use FlashCap again! lol

kekyo, 2023-01-04:
@sappho192 Thanks so much using FlashCap! 
I had the opportunity to shoot images (solely for research) in an MA course student a few years ago, and I wish I had a FlashCap at that time. Because of that time and because I wanted to do the same kind of image shoot in my real work last year, I decided I had to write this myself, so I created it 😵‍💫

Developing on MacOS is a much more difficult situation because I don't have any Mac machine (likewise, iOS...).
If I can get it to work with Android as well, that would be a good thing, since it would cover most of their uses...

sappho192, 2023-01-05:
> Developing on MacOS is a much more difficult situation because I don't have any Mac machine (likewise, iOS...). If I can get it to work with Android as well, that would be a good thing, since it would cover most of their uses...

When I was undergraduate student in 2017, I used [saki4510t/UVCCamera](https://github.com/saki4510t/UVCCamera) to process frame data in Android connected with USB camera. This repository used Android NDK and utilized UVC(USB Video Class).
Even though the repo discontinued after 2017, I think it may be still a good reference for you.

geocine, 2023-01-07:
For 4K preview for a streaming software

kekyo, 2023-01-08:
Thank you for using FlashCap! I know and understand it is slower for transferring large 4K bitstream, related #13 and #6 :)

kekyo, 2023-01-11:
@mlozo Thank you for using! The IMX477 unit looks like embedded purpose, do you use it on RPi with Raspbian OS under armv7l or aarch64?

mlozo, 2023-01-11:
@kekyo I like the (ArduCam B0279) IMX477 cameras for their versatility (and interchangeable lenses), I have a lot of projects where I use Jetsons and RPIs, but here I use the ArduCam B0278 CSI-USB (to USB) adapter and I'm running it on Windows x64.

kekyo, 2023-01-12:
@mlozo Oh, I see that the USB bridge has been released. Can the lenses be interchangeable... I think it could possibly be used for something in my business. Thank you for your information!

gameboycjp, 2023-11-18:
I'm using it in a plugin for a VR game I enjoy. It's extremely niche and will likely only ever be used by a handful of people, but I figured you'd still be interested to know. [KarIO.CaptureDevice](https://github.com/gameboycjp/KarIO.CaptureDevice)

kekyo, 2023-11-19:
@gameboycjp Thanks for reached out! Resonite is a hot topic, I can't imagine how you would use FlashCap, but I hope it fits well.

gameboycjp, 2023-11-20:
@kekyo I have a few ideas in mind. For instance, videos of performances that blur the line between reality and the game. Or, a friend of mine at one point overlayed their headset camera onto the game using OBS, I wanna make that viable in-game.

gentledepp, 2024-05-02:
We are evaluating this library, since we need to port a Xamarin.Forms based application, but MAUI is still too brittle/buggy.
Now, there is an alternative - AvaloniaUI - but that does not support QR Code scanning or taking pictures on Windows devices.

It turns out, that I could modify your sample solution (FlashCap.Avalonia.UI) to easily use zxing.net barcode scanning:

1. Add nuget reference to ZXing.Net.Bindings.ShiaSharp to `FlashCap.Avalonia.UI.csproj`

    ` <PackageReference Include="ZXing.Net.Bindings.SkiaSharp" Version="0.16.14" />`

2. In `MainWindowViewModel`,  add usings on top
```csharp
using ZXing;
using ZXing.SkiaSharp;
```

3.. ... and modify `OnPixelBufferArrivedAsync...`
```diff

+    BarcodeReaderGeneric _zxingReader = new ();

+    public string? ScannedQRCodeText { get; private set; }
    
    private async Task OnPixelBufferArrivedAsync(PixelBufferScope bufferScope)
    {
        ////////////////////////////////////////////////
        // Pixel buffer has arrived.
        // NOTE: Perhaps this thread context is NOT UI thread.
#if false
        // Get image data binary:
        byte[] image = bufferScope.Buffer.ExtractImage();
#else
        // Or, refer image data binary directly.
        ArraySegment<byte> image = bufferScope.Buffer.ReferImage();
#endif
        // Decode image data to a bitmap:
        var bitmap = SKBitmap.Decode(image);

+        var ls = new SKBitmapLuminanceSource(bitmap);
+        var b = _zxingReader.Decode(ls);
        
        // Capture statistics variables.
        var countFrames = Interlocked.Increment(ref this.countFrames);
        var frameIndex = bufferScope.Buffer.FrameIndex;
        var timestamp = bufferScope.Buffer.Timestamp;

        // `bitmap` is copied, so we can release pixel buffer now.
        bufferScope.ReleaseNow();

        // Switch to UI thread:
        if (await UIThread.TryBind())
        {
            // Update a bitmap.
            this.Image = bitmap;

            // Update statistics.
            var realFps = countFrames / timestamp.TotalSeconds;
            var fpsByIndex = frameIndex / timestamp.TotalSeconds;
            this.Statistics1 = $"Frame={countFrames}/{frameIndex}";
            this.Statistics2 = $"FPS={realFps:F3}/{fpsByIndex:F3}";
            this.Statistics3 = $"SKBitmap={bitmap.Width}x{bitmap.Height} [{bitmap.ColorType}]";
            
            
+            if (b is not null && !string.IsNullOrWhiteSpace(b.Text))
+            {
+               ScannedQRCodeText = b.Text;
+            }
+            else
+            {
+                ScannedQRCodeText = "NO CR QODE DETECTED....";
+            }
        }
    }
```

[image]

We are just a bit hesitant, since the microsoft documentation claims DirectShow and Video for Windows are obsolete.
See  [here for DirectShow](https://learn.microsoft.com/en-us/windows/win32/directshow/directshow) and [here for Video for Windows](https://learn.microsoft.com/en-us/windows/win32/multimedia/video-for-windows) :-|

a. do you think they won't be supported anytime soon?
b. do you plan to implement the successor API of them (whatever it is)?

kekyo, 2024-05-02:
Good job! Actually, the very early FlashCap projects started with reading barcodes :)

Yes, I think we are in a delicate situation regarding DirectShow, even if no one is interested in VFW anymore.

I think DirectShow will continue to remain as it is and the API (COM dll library) will remain,
because many Win32 applications still use DirectShow as a backend for AV data operation.

Microsoft encouraged the move to the Media Foundation API and further encouraged the move to UWP (WinRT).
At the end of the day, we need to allow for the fact that problems on DirectShow will not be improved.

It is my personal opinion that these attempts to force the API transition have resulted in a number of projects, both public and private, getting stuck. I think the problem is that these APIs failed to offer any clear advantages other than being up-to-date.

At FlashCap, I think that:

* I would like to support the Media Foundation APIs. However, I cannot take action immediately because I do not have sufficient knowledge (to me) about this API.
* I am not considering support for the UWP capture API; FlashCap is dependency neutral and does not understand how to make this happen in UWP.

The Media Foundation API has been discussed in a previous issue: https://github.com/kekyo/FlashCap/issues/15#issuecomment-1313844748

Both would be happy to accept PR from anyone willing to contribute to this.

gentledepp, 2024-05-06:
> I am not considering support for the UWP capture API
just in case you are curious, the QR Code scanning library I am trying to port to avalonia _is_ actually using that API.

kekyo, 2024-05-07:
Thanks for the info! Actually, the general API interface can hardly be used as is in FlashCap. Because the implementation of namespaces starting with `Microsoft.Maui` depends on MAUI-specific assemblies, and if you implement them as is, you will not be able to observe package dependency neutrality.

FlashCap does not depend on any external packages because it does not use them and achieves this by directly using low-level APIs (both DS, VFW, V4L2, and the Mac port we are currently working on in #45).

The difficulty is how to achieve these without using the APIs generally exposed in UWP and MAUI, and this is where it is unclear.

Perhaps the most foolproof way is to directly import into the internals almost the same definitions as the interfaces exposed by the UWP API. However, UWP itself uses special .NET metadata, so it is unclear whether it is possible to achieve this in such a straightforward way, and the effort involved in checking this is a stumbling block.