kekyo/FlashCap #30: 没有按照 Characteristics 里的 FPS 回调获取画面

yangjieshao, 2022-12-28:
[image]
```

private static async void Start(int selectCameraIndex,int selectCharacteristicsIndex)
{
	var devices = new CaptureDevices();
	var descriptor0 = devices.EnumerateDescriptors().ElementAt(selectCameraIndex);

	Console.WriteLine($"Selected camera :{descriptor0.Name}  Characteristic:{descriptor0.Characteristics[selectCharacteristicsIndex]}");
	using var device = await descriptor0.OpenAsync(
		descriptor0.Characteristics[selectCharacteristicsIndex],
		  bufferScope =>
		{
			Console.WriteLine($"Timestamp:{bufferScope.Buffer.Timestamp}  FrameIndex:{bufferScope.Buffer.FrameIndex}");
		});

	await device.StartAsync();

	Console.ReadKey();

	await device.StopAsync();
}
```
使用的是 1 FPS 的 Characteristic 时间是按照 25 FPS 返回的画面

yangjieshao, 2022-12-28:
[image]
刚才又看了下 获取的图像大小也与设置的 Characteristic 不一致
```

private static async void Start(int selectCameraIndex,int selectCharacteristicsIndex)
{
	var devices = new CaptureDevices();
	var descriptor0 = devices.EnumerateDescriptors().ElementAt(selectCameraIndex);
	bool checkSize = false;

	Console.WriteLine($"Selected camera :{descriptor0.Name}  Characteristic:{descriptor0.Characteristics[selectCharacteristicsIndex]}");
	using var device = await descriptor0.OpenAsync(
		descriptor0.Characteristics[selectCharacteristicsIndex],
		  bufferScope =>
		{
			if(!checkSize)
			{
				checkSize = true;
				byte[] image = bufferScope.Buffer.ExtractImage();

				var ms = new MemoryStream(image);
				var bitmap = Bitmap.FromStream(ms);
				Console.WriteLine($"Size:{bitmap.Size}");
				bitmap.Dispose();
			}

			Console.WriteLine($"Timestamp:{bufferScope.Buffer.Timestamp}  FrameIndex:{bufferScope.Buffer.FrameIndex}");
		});

	// Start processing:
	await device.StartAsync();

	Console.ReadKey();

	await device.StopAsync();

}
```

kekyo, 2022-12-28:
@yangjieshao I have added a function to the Avalonia sample code to display actual FPS measurements, so please try it out in your environment. It may be possible that the camera device you are using is returning lied information.

#31 f3e4f14e1f7eed829e8467f34d40dc8bb0b7be03

The following example is for a device I have. "FPS=nnn/mmm" is the actual value, the first value is able to reach the callback delegate, and the second value is able to reach the FlashCap internals. A smaller difference indicates a smaller of frame drops that can be recognized.

* There is a possibility that there are other frame drops that are not recognized by FlashCap.
* The following tests were done on Windows because Windows was the only environment available for testing right just now. Therefore, it is possible that there may be a problem with V4L2...

### Logicool Webcam C930e

The characteristics on this camera seems accurate.

1920x1080 JPEG 30fps

[image]

1920x1080 YUYV 5fps

[image]

### HD Webcam eMeet C970L

The low FPS characteristics are very suspicious.

1920x1080 JPEG 30fps

[image]

1920x1080 JPEG 1fps

[image]

### Cam Link 4K

FHD/60fps.

[image]

kekyo, 2022-12-28:
In 55bb1743c9185906115ecfe17ec13b2e11af2843, add SKBitmap statistics. Logicool camera is valid.

[image]

kekyo, 2022-12-28:
eMeet C970L 1fps is returned SKBitmap: 640x480

[image]

yangjieshao, 2022-12-29:
thanks
I will buy a new camera for testing
The camera used before is a cheap camera with a price of more than ten yuan. It may indeed return false information

kekyo, 2022-12-29:
Note: I imagine, looking at another existing applications, that sometimes 1fps and/or 5fps are not present in the choices in the selection list, so these may have been excluded from the list as an apparently unusual characteristic by these application.

closed 2022-12-29 by yangjieshao

kekyo, 2022-12-30:
@yangjieshao  If it was obviously strange, please reopen it 😄