kekyo/FlashCap #32: How to capture one image

squirrelfeng, 2023-01-03:
I notice you removed the captureone function, could you please give a sample how to only capture one image？ Thanks!

kekyo, 2023-01-04:
@squirrelfeng I have added a new sample console application called `FlashCap.OneShot` for you to look at. This should suit your needs, as it is simple console-based code without any GUI fragments.

The central design is as follows:

1. Select a device and its characteristics.
2. Open the device and specify a callback. In the callback, use `TaskCompletionSource<byte[]>` to copy the first image data obtained and pass it to subsequent processing.
3. Start capturing, wait for `TaskCompletionSource<byte[]>` to complete, and close it.
4. Save the obtained image data to a file.

kekyo, 2023-01-04:
[Added `TakeOneShotAsync()` in feature/32 branch.](https://github.com/kekyo/FlashCap/blob/d7e203d02964611daa50d1138d7148fb367a7975/samples/FlashCap.OneShot/Program.cs#L62) Did you want a method like this?

squirrelfeng, 2023-01-04:
Thanks! Soo fast response, it worked. Appreciate your work.
If possible, I will suggest to add this function to lib.

closed 2023-01-04 by squirrelfeng

kekyo, 2023-01-04:
@squirrelfeng OK, I will merge to main branch :)

kekyo, 2023-01-04:
1.5.0 is being deployed, contains this feature.