kekyo/FlashCap #36: Linux hangs on OpenAsync and blocks device afterwards

Juff-Ma, 2023-02-19:
await device.OpenAsync() blocks the device and hangs.

Camera itself works normally.

kekyo, 2023-02-21:
Thank you for using FlashCap!

I think, some cameras return unusual characteristic information, which may not work properly if selected. If this case, try selecting a different item of the returned characteristic information.

As a quick way to verify the above, try one of the sample codes at [sample code](https://github.com/kekyo/FlashCap/tree/main/samples) to see if the camera is recognized. If you use the Avalonia sample version, you can use easy the GUI to switch between the cameras and their characteristics.

* These should work if you clone this repository and build FlashCap.sln.
  If the build fails (perhaps due to an older .NET Core SDK not being installed, etc.),
  extract the sample code project, remove the reference to the FlashCap project,
  and instead reference the FlashCap NuGet package.

kekyo, 2023-03-05:
Since there seems to be no response, close it once. Reopen if necessary.

closed 2023-03-05 by kekyo