kekyo/FlashCap #43: V4L2: Couldn't map video buffer: Code=22

dkjonas, 2023-04-09:
I have been using FlashCap 1.5.0 on a Raspberry Pi 4 without any issues.
Now I upgraded the OS from Raspbian 10 (buster) to  Raspbian 11 (bullseye) aka. the newest Raspberry Pi OS.
Unfortunately I'm now unable to connect to my camera.

```
CAM: Arducam OV9782 USB Camera: Ardu: usb-0000:01:00.0-1.3: uvcvideo, Characteristics=25
CAM: 640x480 [JPEG, 30.000fps]
Unhandled exception. System.ArgumentException: FlashCap: Couldn't map video buffer: Code=22, DevicePath=/dev/video0
   at FlashCap.Devices.V4L2Device.<>c__DisplayClass14_0.<OnInitializeAsync>b__0() in /_/FlashCap.Core/Devices/V4L2Device.cs:line 144
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__272_0(Object obj)
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at FlashCap.CaptureDeviceDescriptor.InternalOnOpenWithFrameProcessorAsync(CaptureDevice preConstructedDevice, VideoCharacteristics characteristics, Boolean transcodeIfYUV, FrameProcessor frameProcessor, CancellationToken ct) in /_/FlashCap.Core/CaptureDeviceDescriptor.cs:line 85
   at Core.WebcamAcquisition.SetupAndStart(CancellationToken token) in D:\Source\Core\Core\WebcamAcquisition.cs:line 51
   at Core.Program.Main(String[] args) in D:\Source\Core\Core\Program.cs:line 251
   at Core.Program.<Main>(String[] args)
Aborted
```

I have looked at the source of FlashCap, but that doesn't bring me closer to understand what the error code 22 is about.

Source of simple problem that throws the above exception:
```
            var devices = new CaptureDevices();
            var descriptors = devices.EnumerateDescriptors().
                Where(d => d.Characteristics.Length >= 1).             // One or more valid video characteristics.
                ToArray();

            if (descriptors.ElementAtOrDefault(0) is { } descriptor0)
            {
                var characteristics = new VideoCharacteristics(PixelFormats.JPEG, 640, 480, 30);

                // Show status.
                m_logger.Post(new LogMessage("CAM: " + descriptor0.ToString(), LogMessage.LogLevel.Information));
                m_logger.Post(new LogMessage("CAM: " + characteristics.ToString(), LogMessage.LogLevel.Information));

                m_captureDevice = await descriptor0.OpenAsync(
                    characteristics,
                    OnPixelBufferArrivedAsync);

                // Start capturing.
                await m_captureDevice.StartAsync(token);
            }
            else
            {
                m_logger.Post(new LogMessage("CAM: Device Not found", LogMessage.LogLevel.Error));
            }
```

I tried with different cameras, but that doesn't help. The cameras are working fine in other program on the Pi with the new OS.

Any suggestions?

sappho192, 2023-04-09:
> I have looked at the source of FlashCap, but that doesn't bring me closer to understand what the error code 22 is about.

I think the error code 22 is came from Linux native error code. [Link](https://stackoverflow.com/questions/29890009/linux-system-call-aio-write-fails-with-error-code-22-einval)

Maybe I/O argument for /dev/video is changed since Raspbian 11...?

kekyo, 2023-04-10:
I can't look it up on RPi right now, but what about sappho's point that /dev/video may not exist?

sappho192, 2023-04-11:
I've searched some [[similar](https://github.com/fangfufu/Linux-Fake-Background-Webcam#v4l2loopback)] [[issues](https://stackoverflow.com/questions/70167607/linux-v4l2-driver-video-register-device-return-invalid-parameter)], maybe V4L2 related code needs to be updated. 
But it's odd because Raspbian 11 was released two years ago, and FlashCap was released last year.

> Source of simple problem that throws the above exception:
Trying to get stack trace (toward Linux system calls) with this code may help discover the problem.

dkjonas, 2023-04-11:
> but what about sappho's point that /dev/video may not exist?
Device exits and I can open it with a native linux program (guvcview).
Also, if FlashCap couldn't open the device, then I think it would fail earlier. It does some ioctl on the fd, before the failing mmap.

> I suggest getting a stack trace (toward Linux system calls) with this code may help discover the problem.
I'm not sure how to do this.

dkjonas, 2024-04-05:
Following up on this, just in case others experience the same issue.

The problem is that the kernel on Raspberry Pi4 may be 64 bit, even if user space is 32 bit (see https://github.com/raspberrypi/firmware/issues/1795). Because of this uname will return aarch64 and the 64 bit Interop is used. This makes the mmap call fail.

To get it running on my system, I changed the code in NativeMethods_V4L2.cs to use armv7l Interop for the aarch64 case.

This of course not a general solution, but I have no idea of how to create a better architecture selection.

kekyo, 2024-04-07:
Oh, understood... This difference could also be in Win32 WoW. (Fortunately in Win32 we didn't have to interact directly with the native interface, so the problem didn't occur.)

It might be a good idea to additionally check IntPtr.Size to choose between using aarch64 or armv7l (maybe we should do the same for x86_64).

kekyo, 2024-04-09:
Fixed and will release on next version.

closed 2024-04-09 by kekyo