kekyo/FlashCap #45: Video capturing on Mac devices

YohDeadfall, 2023-04-10:
First of all I would like to thank you for the library. It's really great and has very clean API (: I had an almost complete implementation for capturing V4L2 video, but when I found FlashCap, without any doubt removed the results of several days of my work.

There's only thing missed in my opinion which can be supported too - video capturing on Mac devices. There are examples in the official Apple documentation, but using Objective-C, and bindings can be taken from Xamarin (located in Xamarin.Mac).

If there's nothing other urgent in the project I'm working on, I can contribute it in foreseeable future.

kekyo, 2023-04-11:
Thanks for trying FlashCap!

I'm not making progress with FlashCap right now due to the following issues with Mac support:.

* I don't have a Mac and don't understand the architecture 😄
* (Probably) Need to write native bindings myself.
  * Why: because I want to eliminate references to anything other than .NET BCL from the package dependencies as much as possible.  (You can see this by referring to the current FlashCap package dependency list.  This makes it completely stress-free, as I don't have to think about the consistency of the package with the environment in which it is installed.)
  * I have briefly considered supporting Android, but for the same reason I don't want to make references to Xamarin packages. So I think the way to achieve this would be to structure it as a P/Invoke call to the C endpoint of the Android NDK. This is the reason why FlashCap is currently P/Invoking VFW, DS, and V4L2 all by itself.

Therefore, if a port to Mac is realized, I would welcome it, but I hope it will follow this policy as much as possible.
In the extreme, if it's too much trouble to write from scratch, you could import and inline only the necessary definitions from the binding library (though in that case you would have to pay attention to the copyright).

I myself have not set any specific project progress in FlashCap, so feel free to do so at any time!

kekyo, 2025-03-31:
@ffquintella I would like to write the operating environment in the README, can you please indicate the OSX entry as follows?

```markdown
* Windows (DirectShow devices, tested on x64/x86)
* Windows (Video for Windows devices, tested on x64/x86)
* Linux (V4L2 devices, supported on x86_64/i686/aarch64/armv7l/mips)
```

Other OSX-specific notes, if any, are also noted.

ffquintella, 2025-03-31:
* OSX (AVFoundation devices, supported on x86_64/arm64)

kekyo, 2025-04-01:
Version 1.11.0 package is deploying now. Thanks!

closed 2025-04-01 by kekyo