kekyo/FlashCap #46: FlashCap.Avalonia 拖拽窗口的是视频不播放 (Video does not play when dragging the window)

yangjieshao, 2023-04-16:
在鼠标在标题栏按住不放或者拖拽移动窗口的时候，视频不播放
(When the mouse is held down in the title bar or dragged to move the window, the video will not play)

日志显示，新的图片数据有更新给viewmodel
(The log shows that new image data has been updated to the viewmodel)
```
        // Update a bitmap.
        Dispatcher.UIThread.Post(() =>
        {
            this.Image = bitmap;
            // Update statistics.
            var realFps = countFrames / timestamp.TotalSeconds;
            var fpsByIndex = frameIndex / timestamp.TotalSeconds;
            this.Statistics1 = $"Frame={countFrames}/{frameIndex}";
            this.Statistics2 = $"FPS={realFps:F3}/{fpsByIndex:F3}";
            this.Statistics3 = $"SKBitmap={bitmap.Width}x{bitmap.Height} [{bitmap.ColorType}]";
            Debug.WriteLine("updated a bitmap.");
        }, DispatcherPriority.MaxValue);
```

所以这是 SkiaImageView.Avalonia 的问题？还是Avalonia 的问题？
(So this is the issue with SkiaImageView.Avalonia? Or is it Avalonia's problem?)

[Avalonia.GIF](https://github.com/AvaloniaUI/Avalonia.GIF) 鼠标在标题栏按住不放或者拖拽移动窗口的时候，动画是正常播放的
([Avalonia.GIF](https://github.com/AvaloniaUI/Avalonia.GIF) When the mouse is held down in the title bar or dragged to move the window, the animation plays normally)

yangjieshao, 2023-04-17:
我尝试在获取到新帧画面后刷新纯色背景
(I tried refreshing the solid background after obtaining a new frame image)
[image]
[image]

鼠标在标题栏按住不放或者拖拽移动窗口的时候，背景色未刷新
(When the mouse is held down in the title bar or dragged to move the window, the background color does not refresh)

我在另一个项目中使用类似的代码
(I am using similar code in another project)
[image]
[image]
鼠标在标题栏按住不放或者拖拽移动窗口的时候，背景色刷新正常
(When the mouse is held down in the title bar or dragged to move the window, the background color refreshes normally)

不同之处在于我使用的是 ReactiveUI.Fody
(The difference is that I am using ReactiveUI. Fody)
所以  这会不会是 Epoxy.Avalonia 的错误？
(So could this be Epoxy.Avalonia's mistake?)

yangjieshao, 2023-04-17:
我使用 EventBinder.Avalonia 和 ReactiveUI.Fody 替代了 Epoxy.Avalonia 这个问题就解决了
(use 'ReactiveUI.Fody' and 'EventBinder.Avalonia' to replace 'Epoxy.Avalonia' )
(Solve the problem of the mouse being held down in the title bar and the video not refreshing when dragging the window)
初步怀疑是 Epoxy.Avalonia 的 bug
(Preliminary suspicion is that there is a bug in Epoxy.Avalonia)
https://github.com/kekyo/FlashCap/pull/47

kekyo, 2023-04-17:
@yangjieshao Thanks your report! I have not checked the details yet, but will try to follow the issue from there as it appears to be an Epoxy issue. Please hold this issue as is.

yangjieshao, 2023-05-11:
看上去 这是 Avalonia 的问题
(It seems like this is Avalonia's issues)
https://github.com/AvaloniaUI/Avalonia/issues/11322

closed 2023-05-11 by yangjieshao