kekyo/FlashCap #48: A generic error occurred in GDI+

euklad, 2023-04-23:
I run FlashCap.WindowsForms app on Visual Studio 2022 and see only exceptions that thrown when the image is restored from the stream:

System.Runtime.InteropServices.ExternalException (0x80004005): A generic error occurred in GDI+.
   at System.Drawing.Image.FromStream(Stream stream, Boolean useEmbeddedColorManagement, Boolean validateImageData)
   at System.Drawing.Image.FromStream(Stream stream)
   at FlashCap.WindowsForms.MainForm.OnPixelBufferArrived(PixelBufferScope bufferScope) in C:\repos\temp\FlashCap-main\samples\FlashCap.WindowsForms\MainForm.cs:line 89
   at FlashCap.FrameProcessors.DelegatedQueuingProcessor.ThreadEntry() in C:\repos\temp\FlashCap-main\FlashCap.Core\FrameProcessors\QueuingProcessor.cs:line 159

May be I am doing something wrong?

PS: FlashCap.Avalonia works fine

kekyo, 2023-04-30:
Sorry for the late reply.

I checked in the environment at hand and the problem did not occur. However, in the following line:

```csharp
// Decode image data to a bitmap:
var bitmap = Image.FromStream(stream);
```

is executed on the worker thread called back. Perhaps there is a problem there. For example,

```csharp
    private void OnPixelBufferArrived(PixelBufferScope bufferScope)
    {
        // Force copy before transitive to UI thread context.
        byte[] image = bufferScope.Buffer.CopyImage();

        // `bitmap` is copied, so we can release pixel buffer now.
        bufferScope.ReleaseNow();

        // Switch to UI thread.
        this.synchContext.Post(_ =>
        {
            // Convert to Stream (using FlashCap.Utilities)
            using (var stream = image.AsStream())
            {
                // Decode image data to a bitmap:
                var bitmap = Image.FromStre am(stream);

                // HACK: on .NET Core, will be leaked (or delayed GC?)
                //   So we could release manually before updates.
                var oldImage = this.BackgroundImage;
                if (oldImage != null)
                {
                    this.BackgroundImage = null;
                    oldImage.Dispose();
                }

                // Update a bitmap.
                this.BackgroundImage = bitmap;
            }
        }, null);
    }
```

then `Image.FromStream(stream)` will be executed in the UI thread, which may solve the problem. However, since more bitmaps are copied, frame dropping is more likely to occur.

kekyo, 2023-11-14:
Close this issue. If there is a problem, open again.

closed 2023-11-14 by kekyo