kekyo/FlashCap #49: Generate video

rustam-aytbekoff, 2023-05-02:
Hi!

Is there way to save captured frames as video?
I didn't find anything related to this topic.
Maybe FlashCap has native tools to do that?

For now, I'm saving frames as .jpg files then compiling them into video through FFmpeg.

yangjieshao, 2023-05-02:
> Hi!
> 
> Is there way to save captured frames as video? I didn't find anything related to this topic. Maybe FlashCap has native tools to do that?
> 
> For now, I'm saving frames as .jpg files then compiling them into video through FFmpeg.

https://github.com/secile/OpenH264Lib.NET

you can try this

kekyo, 2023-05-03:
Thank you for using FlashCap!

Yes, FlashCap does not have the ability to output to video format streams (such as mp4 and others) built in. Output to a video stream is a fairly large topic, so I do not believe that it is out of scope to include it into FlashCap.

Since FlashCap can retrieve image data for each frame, it would be possible to further combine this with an external library to encode to a video stream. I have not tried OpenH264Lib.NET, but from a quick read through, it looks easy to implement and is a good option (although real-time may be tough...)

rustam-aytbekoff, 2023-05-03:
Thank you for reply and for the lib you have created.
Yes it seems OpenH264Lib.NET suitable to me.