kekyo/FlashCap #50: Add support for video rotation

roelvdwater, 2023-05-07:
Hi,

I have been using FlashCap for one of my personal projects recently and I'm really happy with it and the performance also seems to be really good from the first impressions. However, one option that I'm really missing is to rotate the video. For some reason my camera delivers the feed with a 90 degrees rotation and I cannot find a way to make this work. Locally (Windows) I have it working using Magick.NET, but on my Raspberry Pi, which is where the app is running on, this doesn't work. This made me wonder if this feature is something that should be built into this library.

I'd like to hear your opinion on this.

kekyo, 2023-05-10:
Thanks for using FlashCap!

So you are saying that the camera you are using (in other applications) can rotate the angle of view? Perhaps that is because of the capture device properties that the camera has that can be manipulated. If this feature is available, I imagine it would be much more efficient than trying to rotate the angle of view after the fact, since the rotation would be hardware-based.

There is currently an attempt in FlashCap to make the capture device properties manipulatable (https://github.com/kekyo/FlashCap/pull/42)
The immediate goal of this PR is to support only DirectShow devices, although I would like to support V4L2 in the future.

By the way, I have not confirmed this, but I think SkiaSharp can also be used to rotate the angle of view. However, SkiaSharp is software-based, even though it is fast, so it may be tough to process in real time.

roelvdwater, 2023-05-11:
Thank you for your reply. I have played around with the camera a little bit yesterday and found out that if you physically rotate the camera, it will also rotate the feed. This means that if you hold it upside down, the feed is upside down as well. In my personal case it fits better to use the camera in portrait mode, although I think that the camera is meant to be used horizontally, as that's how the view looks correct when using the maximum resolution of 1920x1080.

I have tried a number of different libraries, like Magick.NET, ImageSharp and SkiaSharp but none seem to be working on a Raspberry Pi (armv7), which is where I have my webcam connected to, so I was wondering if there is some way to build this into the library "natively".

roelvdwater, 2023-05-16:
Luckily I had another Raspberry Pi that I was able to install the 64-bit version of Raspberry Pi OS on and I can tell you it is working with the `Magick.NET-Q8-arm64` NuGet Package, but the performance is really poor, so I'm going to see what I can do to make the performance better. Once I find a way to achieve this, I'll share the relevant code.

PhenomenCan, 2023-06-16:
> Apparently Magick.NET only has support for the 64-bit version of Raspberry Pi OS. Luckily I had another Raspberry Pi that I was able to install the 64-bit version of Raspberry Pi OS on and I can tell you it is working with the `Magick.NET-Q8-AnyCPU` NuGet Package, but the performance is really poor, so I'm going to see what I can do to make the performance better. Once I find a way to achieve this, I'll share the relevant code.

Hello, were you able to find a performant solution?

roelvdwater, 2023-06-19:
I haven't had time to work on it recently. I might pick this up somewhere in the future when I have some more time hopefully.