kekyo/FlashCap #52: Missing Camera Characteristics

insam125, 2023-05-16:
Hi, 
I did non a lot of research and testing and I am still running into the following issue.

So I have a Logitech Stream Cam connected via USB-C 3.0
The sample Code to print the camera characteristics works fine, but I am missing the 60 FPS seletion (see screenshot)
[image]

For comparison in the windows Camera App there is the option to select 60FPS.
Sorry if I am missing something obvious, but I don't see any solution to this problem.

For my project idea I absolutely need 60 FPS - Any Ideas why the DircetShow pro. for 60 FPS is missing?
Thanks in advance
Kind regards
M.

kekyo, 2023-05-17:
Thank you for using FlashCap!

Maybe this camera supports 60FPS in a format that is not supported by FlashCap.
I did a test branch and made some modifications to FlashCap:

https://github.com/kekyo/FlashCap/tree/feature/52

Check out this branch, open sln and build it, then run the Avalonia sample code.

If you see an item in the format selection combo box that is `Unknown` at 60FPS, then that is a possibility. Selecting that combo box item will raise an exception, but the exception message will contain a value that indicates the format (either an easy-to-understand name or a seemingly obscure value like GUID), so if you can identify this, maybe you can handle it.

I imagine it might be some kind of YUV format that FlashCap does not support (in that case, we would need to implement a transcoder...)

insam125, 2023-05-17:
thanks a lot for the prompt feedback.
After some straggling with .NET Framework 4.0 Dev I was able to check out the feature/52 branch and build it.

the unknown properties are now available (see screenshot) but non of them has the 60 FPS :(
[image]
[image]

For comparison, the windows camera app options:
[image]

Could there be a general problem with DirectShow and Logitech Cameras?

kekyo, 2023-05-17:
Interesting, and unexpected... :(

I suspect some kind of DirectShow issue, since devices that actually support 60FPS, such as the screenshot we  presented, definitely enumerate 60FPS characteristics.
I assume that UWP applications (I believe Windows Camera does too) are internally using the MediaFoundation API, not the DirectShow API.

It could be that the DirectShow API is lying (somehow ignoring that particular), or it could be that the MediaFoundation API is lying (e.g., bitmap doubling to make 30 FPS look like 60 FPS when it is actually 30 FPS).

I have a plan to support the MediaFoundation API (https://github.com/kekyo/FlashCap/issues/15#issuecomment-1313844748), but it has been put on the back burner because it has to be implemented from scratch...

kekyo, 2023-05-17:
If you are willing to follow up, you may be able to isolate the DirectShow issue by trying to see if 60FPS can be selected in an application that can perform video capture with the OSS implementation.

For example, if 60FPS is selectable with OBS, then all you need to do is to check which set of APIs OBS is using. If you are using the MediaFoundation API, then the above hypothesis will be reinforced.

kekyo, 2023-05-26:
Yes, and I am having trouble figuring out what is causing it... What we know at this point is:

* The characteristic information obtained from the DirectShow API has a format `Unknown` that is not supported by FlashCap.
  * This may be NV21, a kind of YUV, since many cameras support NV21.
* The characteristic information obtained from the DirectShow API does not provide a format that supports 60fps.
  * I don't know the reason for this at all. It may be that DirectShow internally drops out formats that do not correspond to the characteristics provided by the camera.
* OBS can select 60fps. From this fact, it is possible that OBS does not use DirectShow API (It uses MediaFoundation API?)

If OBS is using MediaFoundation API, it could work if FlashCap supports MediaFoundation API. We would like to confirm that the MediaFoundation API can reliably acquire 60fps information. However, due to my lack of knowledge of the MediaFoundation API (although I have some documentation on hand in case of such a situation), I have not been able to verify this due to lack of time in the near future...

kekyo, 2024-04-21:
In 1.10.0, supported NV12 transcoder. Maybe it can be improved in YUV format, but I believe it is probably a DS API issue. Give it a try if you are interested. This issue is closed for now.

closed 2024-04-21 by kekyo