kekyo/FlashCap #60: Issue regarding order of known fmtdesc.pixelformat determines if all types are found

AndreSteenbergen, 2023-06-08:
I have a new Logitech C920. This camera breaks the CollectWhile loop, as MJPG is supported. But because it is listed affter the H.264 it is never found.

[image]

```
private static IEnumerable<v4l2_fmtdesc> EnumerateFormatDesc(
        int fd) =>
        Enumerable.Range(0, 1000).
        CollectWhile(index =>
        {
            var fmtdesc = Interop.Create_v4l2_fmtdesc();
            fmtdesc.index = (uint)index;
            fmtdesc.type = (uint)v4l2_buf_type.VIDEO_CAPTURE;
            
            return
                ioctl(fd, Interop.VIDIOC_ENUM_FMT, fmtdesc) == 0 &&
                IsKnownPixelFormat(fmtdesc.pixelformat) ?
                (v4l2_fmtdesc?)fmtdesc : null;
        }).
        ToArray();   // 
```

AndreSteenbergen, 2023-06-08:
I will try and make a PR

AndreSteenbergen, 2023-06-08:
This is my fix:
[image]

AndreSteenbergen, 2023-06-08:
I added the PR, but can't seem to link it ...

kekyo, 2023-06-09:
Thanks for the PR! It looks like you removed `IsKnownPixelFormat()`. This method exists to check if the FlashCap is in a format that will be processed correctly as independent frame data.

Assuming your camera returns `H.264`, is this returned as frame-by-frame independent data? If so, what does the resulting data format actually look like?
I am not familiar with H.264, but I thought H.264 format was continuous (not separated between frames) data, so I thought it should not be included in `IsKnownPixelFormat()`.

I'd be interested to see what the format would look like if I actually made this modification and referenced `BufferScope.Buffer`. Does it say the camera is H.264 but returns it as JPEG or something?

AndreSteenbergen, 2023-06-09:
The enumerator stops at the first signs of null. Not trying the next value anymore.

AndreSteenbergen, 2023-06-09:
H.264 is not frame by frame, so it shoudl retun null. That is good behaviour. Stopping at the first non supported pixelformat is

kekyo, 2023-06-09:
Ouch, I understand it! I will check my environment, thank you explanation.

kekyo, 2023-06-09:
Merged, thank you again!

closed 2023-06-09 by kekyo

AndreSteenbergen, 2023-06-09:
No problem. I ran into trouble with the Logitech C920. You can also add that camera to the supported camera's now ;)