kekyo/FlashCap #63: Support Virtual Webcam

Jelteh, 2023-06-15:
Hello,

Is it possible to connect to a 'Virtual Webcam'? 
I tried the sample but it could not find the OBS Virtual Camera, which I believe is a DirectShow 'device'.

Is this supposed to work or easy to implement?

Sincerely,

kekyo, 2023-06-17:
Thanks for the report. Okay, it does indeed seem that FlashCap does not recognize the OBS virtual camera.
So far, I don't know why it doesn't recognize it, but when I open the DirectShow capture device in VLC, it does recognize it and can actually display it (although sometimes the aspect ratio is wrong), so it doesn't seem that DirectShow cannot recognize it.

It may has unintentionally dropped them out when you are doing the enumeration. I'll check on this in future work (I can't deal with this right away, so if anyone has any insights, I'll accept PRs).

xTektonic, 2026-09-14:
@kekyo 

I looked into this issue when I ran into it about a month ago. The OBS Virtual Camera does not expose the `DevicePath` property which FlashCap uses to identify and enumerate devices. It does, however, have a valid DirectShow moniker display name. My solution was to continue using DevicePath when available and fall back to the moniker for virtual or otherwise pathless devices.

After this fix I ran into another issue, where capture would only produce a single frame before stalling. The device timestamps did not work correctly with the graph’s default reference clock. Disabling that clock for moniker-based sources resolved the problem, while standard sources retain the existing behavior.

My solution is admittedly not the cleanest, but it has worked well for me so I wanted to share what I've learned in hopes that it can help yourself and others who use this library. My implementation can be viewed [here](https://github.com/xTektonic/FlashCap/commit/ba2de264b0c6bcd77c3ad2b65a83963393631e88).