kekyo/FlashCap #93: unable to resolve type SKImageView from namepase

songyitao1991, 2023-09-30:
when I use it in avalonia desktop project , it says unable to resolve type SKImageView from namepase clr-namespace:SkiaImageView;assembly= SkiaImageView 
How to fix it ?  Thanks a lot

kekyo, 2023-10-01:
@songyitao1991 Thanks for using FlashCap!

Is the code you tried in the samples directory of this repository? The code in the `main` branch should be able to run as is, but if you are writing code from scratch, please check and see if you have added the `SkiaImageView.Avalonia` package.

This package allows you to display Skia bitmaps on Avalonia. Note the version of Avalonia you are using: if you are using the Avalonia 0.10 series, you should use the `SkiaImageView.Avalonia` package, if you are using the Avalonia 11 series, the `SkiaImageView. 11` package.

The code in the `main` branch uses the Avalonia 0.10 series; the `develop` branch has been modified to use the Avalonia 11 series for reference.

kekyo, 2023-10-19:
Did you solve the problem? If not, please reopen.

closed 2023-10-19 by kekyo