kekyo/FlashCap #100: need support "loongarch64" Architecture

kakulukiya45t654645, 2023-10-19:


kekyo, 2023-10-19:
Very interesting ;)

I am interested in this architecture, but on the other hand, I live in an area where SBCs for this architecture are not available at all. If you have a porting environment and intend to do it, I can give you some tips on porting, but what do you think?

yangjieshao, 2023-10-20:
https://github.com/dotnet/runtime/tree/main/src/coreclr/pal/src/arch/loongarch64

kekyo, 2023-10-20:
Great, I hope the loongarch64 environment will be readily available.

kekyo, 2023-10-20:
It may be possible to just build it, but I can't confirm that it works on my end. if we want to support the Linux environment, we need to generate source code for the V4L2 bridge.

This may be possible by preparing an environment such as QEMU in the worst case. However, without an environment such as an SBC that works in reality, I cannot check if it works when connected to a camera.

kekyo, 2023-11-16:
I have put together a simplified document for porting FlashCap to the new Linux platform V4L2: https://github.com/kekyo/FlashCap#porting-v4l2-to-unsupported-platforms

If you wish to make your own trial and error, please use this as a reference.

kakulukiya45t654645, 2023-12-13:
@kekyo 你好，我想尝试自己移植到loongarch64架构，但是没怎么看懂说明，请问需要手动编写实现NativeMethods_V4L2_Interop_loongarch64吗

kekyo, 2023-12-13:
No, `NativeMethods_V4L2_Interop_*` is semi-automatically generated using `FlashCap.V4L2Generator` and the included script.

I have written an overview in the document shown in the link. What is not clear?
The work is done in four major steps:

1. Prepare the environment so that Clang etc. can run on the target.
2. Build `FlashCap.V4L2Generator` to work on the target environment.
3. Modify `dumper.sh` to work on the target environment, and execute it. This will generate `NativeMethods_V4L2_Interop_*.cs` file automatically.
4. Modify FlashCap to recognize the generated code.

kakulukiya45t654645, 2023-12-13:
好的，我运行dumper.sh时有报错：error: unknown argument: '-ast-dump=json‘，输出的'videodev2.h.ast.json内容是空的，我在网上没有找到解决方案，请问该如何解决这个问题呢
[image]
[image]

kakulukiya45t654645, 2023-12-13:
这是自动生成的代码文件的内容：
// This is auto generated code by FlashCap.V4L2Generator [1.8.0]. Do not edit.
// Linux version 4.19.0-19-loongson-3 (abuild@10.40.52.160) (gcc version 8.3.0 (Loongnix 8.3.0-6.lnd.vec.36)) #1 SMP 4.19.190.8.14 Thu Aug 24 08:54:20 UTC 2023
// Wed, 13 Dec 2023 09:29:56 GMT

using System;
using System.Runtime.InteropServices;

namespace FlashCap.Internal.V4L2
{
    internal sealed class NativeMethods_V4L2_Interop_loongarch64 : NativeMethods_V4L2_Interop
    {
        // Common
        public override string Label => "Linux version 4.19.0-19-loongson-3 (abuild@10.40.52.160) (gcc version 8.3.0 (Loongnix 8.3.0-6.lnd.vec.36)) #1 SMP 4.19.190.8.14 Thu Aug 24 08:54:20 UTC 2023";
        public override string Architecture => "loongarch64";
        public override int sizeof_size_t => 8;
        public override int sizeof_off_t => 8;

        // Definitions
        public override uint V4L2_CAP_VIDEO_CAPTURE => 1U;
        public override uint V4L2_PIX_FMT_ABGR32 => 875713089U;
        public override uint V4L2_PIX_FMT_ARGB32 => 875708738U;
        public override uint V4L2_PIX_FMT_JPEG => 1195724874U;
        public override uint V4L2_PIX_FMT_MJPEG => 1196444237U;
        public override uint V4L2_PIX_FMT_RGB24 => 859981650U;
        public override uint V4L2_PIX_FMT_RGB332 => 826427218U;
        public override uint V4L2_PIX_FMT_RGB565 => 1346520914U;
        public override uint V4L2_PIX_FMT_RGB565X => 1380075346U;
        public override uint V4L2_PIX_FMT_UYVY => 1498831189U;
        public override uint V4L2_PIX_FMT_XRGB32 => 875714626U;
        public override uint V4L2_PIX_FMT_YUYV => 1448695129U;
        public override uint VIDIOC_DQBUF => 3227014673U;
        public override uint VIDIOC_ENUM_FMT => 3225441794U;
        public override uint VIDIOC_ENUM_FRAMEINTERVALS => 3224655435U;
        public override uint VIDIOC_ENUM_FRAMESIZES => 3224131146U;
        public override uint VIDIOC_QBUF => 3227014671U;
        public override uint VIDIOC_QUERYBUF => 3227014665U;
        public override uint VIDIOC_QUERYCAP => 2154321408U;
        public override uint VIDIOC_REQBUFS => 3222558216U;
        public override uint VIDIOC_S_FMT => 3234878981U;
        public override uint VIDIOC_STREAMOFF => 1074026003U;
        public override uint VIDIOC_STREAMON => 1074026002U;

        // Structures

    }
}

请问是否正常？

kekyo, 2023-12-13:
It seems that Clang does not recognize the option to output ASTs. Perhaps your version of Clang is different than expected.

NOTE: https://stackoverflow.com/questions/59102944/how-to-represent-clang-ast-in-json-format

Some of other information I have seen says to specify `-cc1` instead of `-Xclang`.

kakulukiya45t654645, 2023-12-14:
已经可以在loongarch64架构上运行了，非常感谢
[image]

kekyo, 2023-12-14:
Wow! Congraturations!!
If you may need to merge this patch, please make topic branch on `develop` branch and send PR. I will review and merge soon.

kekyo, 2023-12-19:
Merged, thank you PR and will be released soon.

closed 2023-12-19 by kekyo