kekyo/FlashCap #105: Support for Blackmagic WDM Device

flat-eric147, 2023-11-03:
Hi all,

first of all great library! I experimented a little bit and found an issue with the infinite pixel formats that are out there. I am testing with a BlackMagic SDI input card, their driver includes a DirectShow filter (Blackmagic WDM Capture). It did not work out of the box, so I checked with the debugger and added a fourcc code: 
HDYC = 0x43594448
 which is basically "PixelFormats.UYVY". So adding a few more switches around the code worked for me. 
I forked the repos and pushed my mods here:
https://github.com/flat-eric147/FlashCap
Would be great if this little patch could make it into the main repos!

kekyo, 2023-11-04:
@flat-eric147 Thank you reaching out FlashCap!
 
I don't have BMSDI, but that means the format is fine with UYVY, so I will merge it when I review it!

kekyo, 2023-11-05:
Merged.

closed 2023-11-05 by kekyo