kekyo/FlashCap #108: run wpf sample app

flat-eric147, 2023-11-04:
When compiling and running the FlashCap.Wpf application I get a runtime error:

System.IO.FileLoadException: 'Could not load file or assembly 'System.Runtime.CompilerServices.Unsafe, Version=4.0.4.0

The only way to solve it was to add the package reference (latest version 6.0.0)
System.Runtime.CompilerServices.Unsafe
and update Epoxy.Wpf to the latest version 1.11.0

Happy to push the change if needed.

kekyo, 2023-11-05:
Thanks for getting back to me!

Just to be sure, I made a diff with updated various package references at [feature/109](https://github.com/kekyo/FlashCap/pull/109), could you please try it?

flat-eric147, 2023-11-05:
I just tried and it compiles and runs without issues. Thank you!

kekyo, 2023-11-05:
Merged, thanks!

closed 2023-11-05 by kekyo