kekyo/FlashCap #125: VideoCharacteristics not detected

SignalMQ, 2023-12-22:
**Hello everyone!👋👋👋**
I used an example from the project description. I haven't decided on the characteristics for the video.

**My software and hardware information**

[image]
[image]

**Aditional Context**
```
using FlashCap;

// Capture device enumeration:
var devices = new CaptureDevices();
var descriptor0 = devices.EnumerateDescriptors().ElementAt(0);

byte[] imageData = await descriptor0.TakeOneShotAsync(
    descriptor0.Characteristics.FirstOrDefault());

// Save to file
await File.WriteAllBytesAsync("oneshot", imageData);
```
[image]
[image]

kekyo, 2023-12-25:
Thanks for reached out!

If FlashCap is the latest, `Characteristics` cannot handle are also enumerated as `Unknown` formats. However, I could not understand why they are not enumerated at all, including them. Can we try other cameras?

SignalMQ, 2023-12-26:
**I tried code below:**
```csharp
using FlashCap;
using SkiaSharp;

var devices = new CaptureDevices();
var descriptor0 = devices.EnumerateDescriptors().ElementAt(0);
byte[] imageData = await descriptor0.TakeOneShotAsync(descriptor0.Characteristics.FirstOrDefault());

using (var ms = new MemoryStream(imageData))
{
    SKBitmap bitmap = SKBitmap.Decode(ms);
    SKFileWStream fs = new("photo.png");
    bitmap.Encode(fs, SKEncodedImageFormat.Png, quality: 100);
}
```
**Other cam:**
Logi C270 HD WebCam (USB\VID_046D&PID_0825&MI_00\6&29B00C26&0&0000)
**OS:**
Windows 11 Enterprise 22H2

**Detected and works perfectly...**
**Result:** 
[image]

SignalMQ, 2023-12-26:
> Thanks for reached out!
> 
> If FlashCap is the latest, `Characteristics` cannot handle are also enumerated as `Unknown` formats. However, I could not understand why they are not enumerated at all, including them. Can we try other cameras?

I dont have other cameras on this device, because it's laptop.

kekyo, 2024-01-15:
@SignalMQ On #126 and #127, maybe these issues were a factor. #124 may also affect stable operation, so can I try it after merging these?

SignalMQ, 2024-01-20:
> @SignalMQ On #126 and #127, maybe these issues were a factor. #124 may also affect stable operation, so can I try it after merging these?

I don't understand, what you meaning in 'so can I try it...'. I'm try these fixes or you?

kekyo, 2024-01-20:
Oh, sorry if it doesn't make sense, I made a mistake in the translation program :face_with_spiral_eyes:
Could you try it with a commit from current `develop` branch?

SignalMQ, 2024-01-22:
```csharp

Characteristics [VideoCharacteristics[]]:
{FlashCap.VideoCharacteristics[0]}
Description [string]:
"usb-0000:03:00.4-3: uvcvideo"
DeviceType [DeviceTypes]:
V4L2
Identity [object]:
"/dev/video1"
Name [string]:
"USB2.0 HD UVC WebCam: USB2.0 HD"
Non-Public members
```

kekyo, 2024-02-02:
@SignalMQ Sorry later, I am interested in what characteristics are enumerated in your camera. The following sections:

https://github.com/kekyo/FlashCap/blob/854186a445e494548480ae0d6b607c1bf009eb57/FlashCap.Core/Devices/V4L2Devices.cs#L184C29-L184C48

```csharp
                        return (CaptureDeviceDescriptor)new V4L2DeviceDescriptor(
                            devicePath, ToString(caps.card), $"{ToString(caps.bus_info)}: {ToString(caps.driver)}",
                            EnumerateFormatDesc(fd).
                            SelectMany(fmtdesc =>
                                EnumerateFrameSize(fd, fmtdesc.pixelformat).
                                SelectMany(frmsize =>
                                    EnumerateFramesPerSecond(fd, fmtdesc.pixelformat, frmsize.Width, frmsize.Height).
                                    Collect(framesPerSecond =>
                                        NativeMethods_V4L2.CreateVideoCharacteristics(
                                            fmtdesc.pixelformat, frmsize.Width, frmsize.Height,
                                            framesPerSecond.Value, ToString(fmtdesc.description),
                                            frmsize.IsDiscrete && framesPerSecond.IsDiscrete)))).
                            Distinct().
                            OrderByDescending(vc => vc).
                            ToArray());
```

Here, if a series of `SelectMany()` returns no result, the number of characteristics is zero. Such a possibility would be if `EnumerateFormatDesc()`, `EnumerateFrameSize()`, or `EnumerateFramesPerSecond()` returned zero elements. Your camera may fall into one of these cases.

This line is written on a single line, so it may be difficult to verify. For example:

```csharp
var a = EnumerateFormatDesc(fd).ToArray();
var b = a.SelectMany(fmtdesc =>
{
    var c = EnumerateFrameSize(fd, fmtdesc.pixelformat).ToArray();
    var d = c.SelectMany(frmsize =>
    {
        // ...
    });
    // ...
});
```

to change the enumeration so that you can check the enumeration and be sure of it on the debugger.

SignalMQ, 2024-02-06:
@kekyo I'm tried to detect my device with `V4L2Devices().EnumerateDescriptors()` and suprised for detected 2nd shadow copy of device placed in `/dev/video0`:

[image]

Here it seemed strange to me that the devices are listed in reverse order _([/dev/video1, /dev/video0])_, or I’m wrong.

kekyo, 2024-02-07:
I was surprised too! I'm not sure why there are multiple `/dev/video?` enumerations in the first place, but I'm not particularly aware of the order:

https://github.com/kekyo/FlashCap/blob/854186a445e494548480ae0d6b607c1bf009eb57/FlashCap.Core/Devices/V4L2Devices.cs#L171

If you select `/dev/video1`, will it capture correctly?

SignalMQ, 2024-02-07:
> I was surprised too! I'm not sure why there are multiple `/dev/video?` enumerations in the first place, but I'm not particularly aware of the order: [...]

`/dev/video1` - fail
`/dev/video0` - sucess

kekyo, 2024-02-07:
According to the screenshot, `/dev/video0` has zero characteristics and `/dev/video1` enumerates several characteristics, did you succeed in the case of `/dev/video0`?

Are you saying that you used the value of `/dev/video1` as the characteristic in that case?

SignalMQ, 2024-02-07:
> According to the screenshot, `/dev/video0` has zero characteristics and `/dev/video1` enumerates several characteristics, did you succeed in the case of `/dev/video0`?
> 
> Are you saying that you used the value of `/dev/video1` as the characteristic in that case?

`/dev/video0` has 11 characteristics and it's work, but `/dev/video1` enumerated in first is not work, because don't have any characteristics listed.

kekyo, 2024-02-07:
I see. As a coping remedy, it seems that we need to ignore devices with zero characteristics.

We could ignore devices with zero characteristics when internal FlashCap enumerates them, but I imagine that would probably make it difficult to identify the problem.
Therefore, I would like to make it a policy that this exclusion is handled by the caller. For example:

```csharp
if (devices.EnumerateDescriptors().
    FirstOrDefault(d => d.Characteristics.Any(c => c.PixelFormat != PixelFormats.Unknown)) is { } device)
{
    // `device` is valid.
}
```

SignalMQ, 2024-03-05:
My problem solved, I use my second device.

closed 2024-03-05 by SignalMQ

kekyo, 2024-04-19:
@SignalMQ The next version will support NV12 format. Perhaps devices that were not enumerated (or were Unknown) will become available, so if you are interested, please give it a try.