kekyo/FlashCap #126: Typo in Linux device enumeration

gplwhite, 2024-01-08:
Hello, 

I think there are a couple of mistakes in the enumeration of devices on Linux. There appear to be two places where each DefactoryStandardResolution frame width is incorrectly compared to the enumerated frame max_height - instead of the frame max_width.

https://github.com/kekyo/FlashCap/blob/854186a445e494548480ae0d6b607c1bf009eb57/FlashCap.Core/Devices/V4L2Devices.cs#L66
and 
https://github.com/kekyo/FlashCap/blob/854186a445e494548480ae0d6b607c1bf009eb57/FlashCap.Core/Devices/V4L2Devices.cs#L80

gplwhite, 2024-01-09:
I think there is a second issue with the filtering of the DefactoStandardResolutions.

The following lines don't seem to take operator precedence into account correctly. The modulo division has a higher order of precedence than the subtraction meaning it will happen before the subtraction. I think there should be parenthesis around the subtraction to ensure it occurs before the modulo division.

https://github.com/kekyo/FlashCap/blob/854186a445e494548480ae0d6b607c1bf009eb57/FlashCap.Core/Devices/V4L2Devices.cs#L67

and 

https://github.com/kekyo/FlashCap/blob/854186a445e494548480ae0d6b607c1bf009eb57/FlashCap.Core/Devices/V4L2Devices.cs#L70

Currently I believe this will result in no characteristics being returned for devices that report a ```Stepwise``` frame size capability

(specifically this issue along with https://github.com/kekyo/FlashCap/issues/127 seems to be stopping my Raspberry PI onboard camera from being enumerated successfully)

kekyo, 2024-01-10:
@gplwhite I will review it in a couple of days, just wait and see.

kekyo, 2024-01-15:
Merged, thanks!

closed 2024-01-15 by kekyo