kekyo/FlashCap #127: Reversed logic when enumerating Linux device capabilities

gplwhite, 2024-01-09:
I've been trying to diagnose why I cannot get Flashcap to correctly enumerate a built-in camera on a Raspberry PI (model 3 B+). 

During my investigation I think I've discovered a flaw in the logic for enumerating the device capabilities - specifically related to the enumeration of the FramesPerSecond.

As commented in the code, while the V4L2 api reports the value as a time interval, Flashcap inverts this value to a FramesPerSecond value.
The minimum and maximum intervals are converted to their FPS value and used to filter a list of standardised frame rates.

However, the problem is that while the interval values have been inverted, the minimum and maximum values are still treated as a minimum and maximum when filtering the standardised rates. But the minimum and maximum meanings should also be swapped.

For example, if the interval values reported by the V4L2 api is:
```
min: 10/1
max: 30/1
```
this would be converted to FPS values of
```
min: 1/10
max: 1/30
```

1/10 is a larger value than 1/30 so the minimum and maximum meanings should be reversed:
```
min: 1/30
max: 1/10
```

Because the minimum value is larger than the maximum value, the `fps >= min && fps <= max` filtering constraint can never be true and no FramesPerSecond values are returned.

gplwhite, 2024-01-09:
This will affect devices that report framerates as either ```Stepwise``` and ```Continuous``` values. Devices that report ```Discrete``` values will not be affected since the filtering is not performed.

kekyo, 2024-01-10:
Thank you send PR, I'll review it in a couple of days!

kekyo, 2024-01-15:
Thank you, merged!

closed 2024-01-15 by kekyo