kekyo/FlashCap #132: Elgato CamLink 4K DirectShow Pixel Format

Muny, 2024-02-04:
Before I describe the issue, I want to say that I very much appreciate your effort on this project. It's been extremely useful, and it's so helpful that it's cross-platform.

Now to the issue:

I see that the Elgato CamLink 4K is on the list of verified devices, however it seems to not be working for me.

I'm developing on Win11 with .NET 8.0.

The device enumerates as a DirectShow device with one characteristic:
3840x2160 [Unknown, 29.970fps]

It seems FlashCap doesn't support the pixel format reported by the device. 
Value of RawPixelFormat:
`3231564e-0000-0010-8000-00aa00389b71`

Looking at https://gix.github.io/media-types, it seems this format is a YUV variant called NV12.
(side note: I found this sample code for converting NV12 to RGB: https://paulbourke.net/dataformats/nv12)

I'm curious, when the CamLink 4K was tested before, what pixel format did it report?

kekyo, 2024-02-05:
@Muny Thanks reached out FlashCap!

Hmm, I just tried it in my environment (Ubuntu 22.04 x64 and ), and it is treated as `YUYV` (In `RawPixelFormat`).

[image]

[image]

I don't understand why only `NV12` is listed. In fact, `NV12` is a widely available format and I would like to support it in FlashCap, but it is not supported in the current version.

Maybe it's because I only tested with FHD in my environment. In your report, it is 4K resolution, and if it is 4K resolution, it may only be able to select `NV12`.

Muny, 2024-02-05:
Thanks for the quick feedback. It seems you're right about the resolution the device is outputting having an effect on the reported formats from the Cam Link.

I changed my device to output 1080p video, and I now have a 4:2:2 YUYV characteristic (along with the unsupported NV12):
```
Cam Link 4K: Cam Link 4K (DirectShow), Characteristics=2
1920x1080 [YUYV, 59.940fps], RawPixelFormat: 32595559-0000-0010-8000-00aa00389b71 (YUYV 4:2:2 Packed 16pp 8bpc)
1920x1080 [Unknown, 59.940fps], RawPixelFormat: 3231564e-0000-0010-8000-00aa00389b71 (NV12 4:2:0 Planar 12bpp 8bpc)
```

kekyo, 2024-02-07:
Related #3

kekyo, 2024-04-19:
@Muny Thanks for the long wait. I will be releasing a version with this fix this weekend!

closed 2024-04-19 by kekyo