kekyo/FlashCap #135: How to do shrink memory when capturing from WebCam to WPF

fiery9711, 2024-02-28:
Hi everyone, I'm new user this library, thanks for all who made it. But i have problem when i use webcam and set image to WPF my program size grows up to 2-3GB RAM memory how i can solve this problem?

flat-eric147, 2024-02-28:
I use FlashCap with WPF and I don't see such behavior. Being .net, perhaps it's not doing garbage collection, you might try to trigger the [garbage collector](https://learn.microsoft.com/en-us/dotnet/api/system.gc.collect?view=net-8.0#System_GC_Collect) explicitly and see if the situation improves. Perhaps also share some of your code? How does the WPF sample behave?

kekyo, 2024-04-19:
Related #138

This fix will pool the buffers used by PixelBuffer. Using this specification and `ReferImage()` may improve on situations where GC occurs continuously.

closed 2024-04-21 by kekyo