kekyo/FlashCap #138: AutoPixelBufferScope set PixelBuffer to null, cause allocate new PixelBuffer for every frame

RyanYANG52, 2024-03-21:
Great project, thanks for the work!

I am looking for a windows camera capture lib to use in my next project, your work is perfect. By testing it for two days, I come across some issues, hope you can help take a look at it.

My program record multiple camera at one, so I am a lot strict about GC pressure, try to avoid all unnecessary allocation.

I test with a simple Console app, do nothing in the `pixelBufferArrived`, and I can see `Allocated: CurrentSize=0, Size=282346` in my Debug console for every frame. So I trace to those two line:
https://github.com/kekyo/FlashCap/blob/854186a445e494548480ae0d6b607c1bf009eb57/FlashCap.Core/FrameProcessor.cs#L114

https://github.com/kekyo/FlashCap/blob/854186a445e494548480ae0d6b607c1bf009eb57/FlashCap.Core/PixelBufferScope.cs#L37

If I not mistaking, after `pixelBufferArrived`, `AutoPixelBufferScope` will dispose itself,  then call `base.OnReleaseNow();`, on base method it will set `this.Buffer = null`, then `this.parent.ReleasePixelBuffer(this.Buffer);` will add a null to Stack, cause on next frame stack pop a null Buffer, then `new Buffer`, `CopyIn`, `new byte[size]`

I comment the line `base.OnReleaseNow();` , test in `FlashCap.Wpf` demo project, it works fine, and no new byte[] happen except for first few frames.

I think this maybe a bug.

Another thing I noticed is the use of `new byte[totalSize]` in `CopyIn`. A mjpeg feed every frame size is different a little, so every time a big frame comes, It has to allocate a new byte array, in the consumer side, we only works with ArraySegment, so maybe some optimisation can be done. My thoughts: 

1. Add a IDispose interface to PixelBuffer, rent byte array from ArrayPool, if size too small, return and rent again, return on Dispose
2. Call Dispose when `FrameProcessor.Dispose` or `reserver` is full

Best regards

RyanYANG52, 2024-03-21:
I am writing down my approach for reference.

I am writing a windows wpf app, using WriteableBitmap to render, and ffmpeg api to record.

I decode jpeg using ffmepg from webcam into a BRGA VideoFrame, then in UI thread copy into bitmap. 

using WriteableBitmap is to avoid `new Bitmap` or `SKBitmap.Decode` allocation, in the same camera feed, every frame after decode is the same size.

kekyo, 2024-03-22:
Thanks reported issue!

> I comment the line base.OnReleaseNow(); , test in FlashCap.Wpf demo project, it works fine, and no new byte[] happen except for first few frames.
> I think this maybe a bug.

You are right, this is a bug. We will fix it in the next version!

> Another thing I noticed is the use of new byte[totalSize] in CopyIn.

Yes, MJPEG returns almost different sizes for each frame. I had given up on this because if I cached all of these, a significant amount of arrays would continue to be kept. Certainly this is not a problem when returning an `ArraySegment`, so it may be possible to make caching work depending on the method of acquisition. I'll give it some thought.

closed 2024-04-12 by kekyo