kekyo/FlashCap #146: Support for camera module/ CSIconnector on Pi Zero 2W

ctacke, 2024-04-21:
I am trying to get this up and running on a Raspberry Pi Zer 2W.  I can enumerate the capture devices, and it find 6.  All of the devices, however, report zero characteristics, so I can't open them for frame capture.

```
var devices = new CaptureDevices();
if (devices == null)
{
    Console.WriteLine($"There are no capture devices found");
    return;
}

var en = devices.EnumerateDescriptors();
Console.WriteLine($"There are {en.Count()} capture devices");

foreach (var descriptor in en)
{
    Console.WriteLine("---");
    Console.WriteLine($" {descriptor.Name}");
    Console.WriteLine($" {descriptor.Description}");
    Console.WriteLine($" {descriptor.DeviceType}");
    if (descriptor.Characteristics.Length == 0)
    {
        Console.WriteLine($" descriptor has zero characteristics");
    }
    else
    {
        foreach (var c in descriptor.Characteristics)
        {
            Console.WriteLine($"   {c.Width}x{c.Height} @ {c.PixelFormat}");
        }
    }
}
```
Gives an output of
```
There are 6 capture devices
---
 unicam
 platform:3f801000.csi: unicam
 V4L2
 descriptor has zero characteristics
---
 unicam
 platform:3f801000.csi: unicam
 V4L2
 descriptor has zero characteristics
---
 bcm2835-isp
 platform:bcm2835-isp: bcm2835-isp
 V4L2
 descriptor has zero characteristics
---
 bcm2835-isp
 platform:bcm2835-isp: bcm2835-isp
 V4L2
 descriptor has zero characteristics
---
 bcm2835-isp
 platform:bcm2835-isp: bcm2835-isp
 V4L2
 descriptor has zero characteristics
---
 bcm2835-isp
 platform:bcm2835-isp: bcm2835-isp
 V4L2
 descriptor has zero characteristics
```

kekyo, 2024-04-23:
Thanks reached out!

I am not sure where the problem lies, as I do not have a camera that connects via MIPI.

Maybe some hack is needed to enumerate V4L2 in RPi's MIPI connection. I found some information about this on a Japanese blog, but I could not find any technical background.

Link for reference: https://syurecat.hatenablog.com/entry/2022/09/11/230339

Oblivionburn, 2024-05-15:
You might have to enable LegacyCamera in raspi-config and install the following libraries in order for .NET Core to fully interact with the camera hardware:
sudo apt-get install v4l-utils
sudo apt-get install libc6-dev libgdiplus libx11-dev

Example project of someone getting a .NET Core project working with a Raspberry Pi 3's camera (different lib, but maybe similar dependency requirements):
https://neilsnotes.net/Software/Coding/dotnetPiCam.html

kekyo, 2024-05-16:
Thanks provided infos!