kekyo/FlashCap #147: OnPixelBufferArrived can not get new YUYV buffer in  v1.10.0  （v1.9.0 is ok）

yangjieshao, 2024-05-22:
when videoCharacteristic.PixelForma  is YUYV

打断点查看数据的话，一切正常
(If you view data in breakpoint, everything is ok)
但是一但正常运行
(But once it runs normally)

在 1.9.0 中一切正常
(everything is ok in 1.9.0)

在1.10.0中 每次获取的数据都是旧数据（事实上每次获取的都是第一帧数据）
(In 1.10.0, each data obtained is old data) [(In fact, every time we obtain the first frame of data)]

this is my sample
[ConsoleApp1.zip](https://github.com/kekyo/FlashCap/files/15397313/ConsoleApp1.zip)

```
private static ArraySegment<byte> _lastImageBuffer;
private static string _lastConsoleMsg = string.Empty;
private static readonly string sameMsg = "same image buffer";
private static readonly string newMsg = "new image buffer";
private static void OnPixelBufferArrived(PixelBufferScope bufferScope)
{
	var imageBuffer = bufferScope.Buffer.ReferImage();
	if (EqualityComparer<ArraySegment<byte>>.Default.Equals(_lastImageBuffer, imageBuffer))
	{
		Debug.WriteLine(sameMsg);
		if(_lastConsoleMsg!= sameMsg)
		{
			Console.WriteLine(sameMsg);
			_lastConsoleMsg = sameMsg;
		}
	}
	else
	{
		_lastImageBuffer = imageBuffer;
		Debug.WriteLine(newMsg);
		if (_lastConsoleMsg != newMsg)
		{
			Console.WriteLine(newMsg);
			_lastConsoleMsg = newMsg;
		}
	}
}
```

yangjieshao, 2024-05-22:
1.10.0
[image]

1.9.0
[image]

Muny, 2024-05-22:
Can you print the list of descriptors and characteristics you discover on your system on v1.9.0 and v1.10.0 using:
```csharp
using FlashCap;

// Capture device enumeration:
var devices = new CaptureDevices();

foreach (var descriptor in devices.EnumerateDescriptors())
{
    // "Logicool Webcam C930e: DirectShow device, Characteristics=34"
    // "Default: VideoForWindows default, Characteristics=1"
    Console.WriteLine(descriptor);

    foreach (var characteristics in descriptor.Characteristics)
    {
        // "1920x1080 [JPEG, 30fps]"
        // "640x480 [YUYV, 60fps]"
        Console.WriteLine(characteristics);
    }
}
```

yangjieshao, 2024-05-22:
> Can you print the list of descriptors and characteristics you discover on your system on v1.9.0 and v1.10.0 using:
> 
> ```cs
> using FlashCap;
> 
> // Capture device enumeration:
> var devices = new CaptureDevices();
> 
> foreach (var descriptor in devices.EnumerateDescriptors())
> {
>     // "Logicool Webcam C930e: DirectShow device, Characteristics=34"
>     // "Default: VideoForWindows default, Characteristics=1"
>     Console.WriteLine(descriptor);
> 
>     foreach (var characteristics in descriptor.Characteristics)
>     {
>         // "1920x1080 [JPEG, 30fps]"
>         // "640x480 [YUYV, 60fps]"
>         Console.WriteLine(characteristics);
>     }
> }
> ```

used same device same videoCharacteristic in 1.9.0 and 1.10.0
800x448 [YUYV,25fps]
[image]

[image]

kekyo, 2024-05-22:
I haven't followed it in detail yet, but maybe the instances are identical, but the data inside is changing?

It is possible that `EqualityComparer<ArraySegment<byte>>.Default.Equals()` is not inspecting the contents of the array `byte[]` inside `ArraySegment<byte>`. I assume this is because the buffer pool was introduced in 1.10.0, so the same array instance is returned.

yangjieshao, 2024-05-22:
It seems you are right