kekyo/FlashCap #148: Support

marakhand, 2024-05-31:
I'm trying to connect to a VHS to Digital Converter device [(like this device)](https://www.amazon.com/DIGITNOW-Capture-Adapter-Grabber-Converter/dp/B08RMSJNW2/ref=sr_1_1?dib=eyJ2IjoiMSJ9.SZW6UN87Os2FqxzSQgm1eWUrXC4PGWqCSg941qZ69PKcLxylO0TdE7OqZrr2SvnL84qFRzXQtIb_BMBtvEvVbN7Ba4LORzqWd6Pvdnuz-hMK5d6ljn3hy2DNP4Uui_66o27Nw9y6fbTVD_Zk4oj-URHd5FjSUUzv9Moz3iTLRQsKRZf1lyWmNw91T2goNfnHOCw7n7W0uCrJGo3CxnYKXOKifTY4J3SPqh14wmATm_4.VEtn74FsiB1pkPeyhhYrOLfRMPhbTFtb6eK5bMN3F0Y&dib_tag=se&keywords=easycap&qid=1717141300&sr=8-1&th=1) on a Linux. 
The device is visible, but there are no characteristics.

`
foreach (var dev in devices.EnumerateDescriptors())
{
    Console.WriteLine($"{dev.Name}, {dev.Description}, length={dev.Characteristics.Length}");
}
`

[image]

[image]

[image]

marakhand, 2024-06-04:
After a little research, it turned out that this is happening because getting VIDIOC_ENUM_FRAMEINTERVALS does not work.
How can I solve this problem?

kekyo, 2024-06-04:
https://github.com/kekyo/FlashCap/blob/227cca10f66e504cf3abb690fb15fe854da16825/FlashCap.Core/Devices/V4L2Devices.cs#L127

You mean that this `ioctl` always gives error?

marakhand, 2024-06-04:
Yes.
The line 150 (Console.Write) is unreachable.

[image]

WLW-CHN, 2025-09-04:
Has this issue been resolved?

kekyo, 2026-11-09:
It's a bit late, but if you're still interested, please try fixing it in the next release (1.13.0).

closed 2026-11-09 by kekyo