kekyo/FlashCap #149: 当同时连接多个摄像头（比如6个）的时候，数据返回会缺失 (When multiple cameras are connected simultaneously (such as 6), data return will be missing)

yangjieshao, 2024-06-04:
我尝试同时链接并播放6路摄像头
(I tried to link and play 6 cameras simultaneously)
每路摄像头是1080p的
(Each camera is 1080p)
在 `OnPixelBufferArrived `打日志看
(Check logs in 'OnPixelBufferArrived')
大概只有1~3路摄像头有视频数据返回
(Approximately only 1-3 cameras have video data returned)
有时候是只有1路，有时候是有3路
(Sometimes there is only one camera data, sometimes there are three camera data)
并且每路摄像头返回的帧序号也都不一样
(And the frame number returned by each camera is also different)
从帧序号上看，有的返回数据快，有的返回慢
(From the perspective of frame numbering, some return data quickly while others return data slowly)

已经使用 `ThreadPool.SetMinThreads` 将线程池开到最大
(ThreadPool has already been used SetMinThreads ` Set thread pool to maximum size)
CPU 占用只有 20%~30%
(CPU usage is only 20%~30%)

[image]

yangjieshao, 2024-06-04:
[image]

kekyo, 2024-06-10:
I can't imagine the problem by looking at the contents, so I need more detailed analysis, but I can't reproduce the situation directly because there is no equipment that can connect so many capture devices here.

If anyone has any suggestions, I would appreciate it.