kekyo/FlashCap #152: How to use for stereo cameras

xuyiiii, 2024-11-04:
[image]

My camera is a binocular camera with both IR and RGB cameras, and I need to enable both cameras simultaneously. But when I turn on the RGB camera then turn on the IR camera, it will return "FlashCap: Could't start capture: Code=28, DevicePath=/dev/video2". My operating environment is Linux+ARM64
How can I solve this problem？ありがとうございます

我的摄像头是双目摄像头，有IR和RGB两个摄像头，我需要同时启用这两个摄像头。但是当我开启RGB摄像头后，开启IR摄像头，就会返回“FlashCap: Couldn't start capture: Code=28, DevicePath=/dev/video2”。我的运行环境是Linux+ARM64。
我该如何解决？谢谢

xuyiiii, 2024-11-08:
I know why now. Initially, I selected the highest resolution of 2592x1944 for both the RGB and IR cameras. 
Testing with another SDK returned "No space left on device." When both cameras were set to a resolution of 640x480, they worked normally. 
Error code 28 might indicate that there is not enough memory for the cameras to open, so the resolution needs to be reduced. 
I hope this helps you.

我知道为什么了。一开始，RGB和IR摄像头我选的都是最高分辨率2592x1944输出。
用别的sdk测试返回"No space left on device"。当两个摄像头都选择640x480输出时，就能正常工作。
错误代码28可能是指：摄像头内存不够无法打开，需降低分辨率。
希望对您有帮助。

closed 2024-11-08 by xuyiiii

kekyo, 2024-11-08:
Thanks wrap up!