kekyo/FlashCap #154: Can I castomize (create my own) property page?

orangebobolink, 2025-01-03:
I use Avalonia UI and want to use theme (or create my own property page) but I haven't found a way of achieving it. Is it impossible to achieve?

[image]

kekyo, 2025-01-03:
Unfortunately, since the properties page uses Win32's native dialogs, the AvaloniaUI theme would not be applicable...

kekyo, 2025-01-03:
I think the answer was a little unclear.

The property pages are displayed by the DirectShow API itself, which uses native Win32 dialogs to display them, and FlashCap cannot get involved in the native dialogs, it can only display them. Presumably, AvaloniaUI also cannot hook into the native Win32 dialogs and apply the theme system.

orangebobolink, 2025-01-03:
Thanks for the quick answear

closed 2025-01-03 by orangebobolink