kekyo/FlashCap #158: Missing Format Support in FlashCap.Core/Internal/V4L2/NativeMethods_V4L2_Interop_loongarch64.cs Causes Crash on Hardware Insertion

leihao, 2025-03-14:
When specific hardware is inserted, the application crashes due to missing format support in FlashCap.Core/Internal/V4L2/NativeMethods_V4L2_Interop_loongarch64.cs. This issue needs to be addressed to ensure stability and compatibility with the hardware.
Crash Log:
```
Unhandled exception. System.TypeInitializationException: The type initializer for 'FlashCap.Internal.NativeMethods_V4L2' threw an exception.
 ---> System.NotImplementedException: The method or operation is not implemented.
   at FlashCap.Internal.V4L2.NativeMethods_V4L2_Interop.get_V4L2_PIX_FMT_NV12() in /home/codingSpace/Desktop/git/FlashCap-main/FlashCap-main/FlashCap.Core/Internal/V4L2/NativeMethods_V4L2_Interop.cs:line 112
   at FlashCap.Internal.NativeMethods_V4L2..cctor() in /home/codingSpace/Desktop/git/FlashCap-main/FlashCap-main/FlashCap.Core/Internal/NativeMethods_V4L2.cs:line 64
   --- End of inner exception stack trace ---
   at FlashCap.Internal.NativeMethods_V4L2.open(String pathname, OPENBITS flag)
   at FlashCap.Devices.V4L2Devices.<OnEnumerateDescriptors>b__8_0(String devicePath) in /home/codingSpace/Desktop/git/FlashCap-main/FlashCap-main/FlashCap.Core/Devices/V4L2Devices.cs:line 194
   at System.Linq.EnumerableExtension.Collect[T,U](IEnumerable`1 enumerable, Func`2 selector)+MoveNext() in /home/codingSpace/Desktop/git/FlashCap-main/FlashCap-main/FlashCap.Core/Internal/BackwardCompat.cs:line 72
```

kekyo, 2025-03-14:
See #153 , we need an environment that will allow us to successfully build loongarch64.
Could you make a PR that includes the fix?

leihao, 2025-03-14:
> See [#153](https://github.com/kekyo/FlashCap/pull/153) , we need an environment that will allow us to successfully build loongarch64. Could you make a PR that includes the fix?

I've submitted a PR to address this issue. [#159](https://github.com/kekyo/FlashCap/pull/159).

kekyo, 2025-03-14:
I think I wrote it badly.

To accept the fix, you need to create a script that can build the loongarch64 build environment in qemu.

The other V4L2 environments we currently support can be built using qemu. These use official Debian images, so you can create an environment that can be reproduced anywhere, anytime.

On the other hand, loongarch64 is not provided as an official Debian image, so I don't know how to build a qemu environment. Also, it is difficult to prepare a build environment for support because the physical hardware is also difficult to obtain.

leihao, 2025-03-14:
I understand now, we need a build environment instead of directly updating the Interop files. 
I am also new to Linux development; most of my coding is done on Windows, with final compilation and debugging on a physical machine. I am not sure what is needed for a QEMU environment. 
Previously, when seeking loongarch64 support for another project, it was also mentioned that a virtual environment was needed. At that time, I provided a UOS image but have not received a response. Regarding the image, maybe you can try Loongnix? Their website is https://pkg.loongnix.cn/loongnix/ and https://github.com/loongson might also be helpful.