kekyo/FlashCap #161: Image corruption issues encountered in different OSX environments

kekyo, 2025-04-01:
With version 1.11.0, FlashCap was ported to OSX. During that work, we discovered a problem with the resulting images being corrupted in different OSX environments.

We made this issue because we could not reproduce the problem in the contributor's environment in the port to OSX.
If anyone can analyze and fix the problem in detail, please write here. It would be helpful.

The issue can be found at [PR #160](https://github.com/kekyo/FlashCap/pull/160#issuecomment-2756455916).

Related: #45

ffquintella, 2025-05-28:
I think my last PR solves this (there was a problem setting the image size) wich is solved now.

kekyo, 2025-05-29:
Thank you! I'm in a bit of a work jam right now, so could you give me a couple of days?

ffquintella, 2025-05-29:
Shure take your time.

kmorwood, 2025-12-17:
I'm just latching onto this as we're starting to use FlashCap. I have multiple macs to test on...and a product in development that shows some interesting image issues. Let me get settled into the repo and I'll start testing.